cmsis-svd/cmsis-svd-data #4: Preserve history from `cmsis-svd/cmsis-svd`

gschwaer, 2024-02-09:
The move of the `data/` folder from [cmsis-svd/cmsis-svd](https://github.com/cmsis-svd/cmsis-svd) to [cmsis-svd/cmsis-svd-data](https://github.com/cmsis-svd/cmsis-svd-data) dropped the commit history.

I tinkered a little and found a way to preserve commit history of all commits that changed files in `data/` (using [git-filter-repo](https://github.com/newren/git-filter-repo)). I cannot open a PR because github won't let me (due to rewriting the history). But here's the script to do the changes that I did in [gschwaer/cmsis-svd-data](https://github.com/gschwaer/cmsis-svd-data):
```bash
#!/usr/bin/env bash
set -eu

git clone git@github.com:cmsis-svd/cmsis-svd.git
cd cmsis-svd

# Discard all commits after (and including) the move.
git reset --hard ":/Remove data directory."
git reset --hard HEAD^

# Keep all commits that changed `data/`.
git filter-repo --path data/ --force

# Update the merge commit messages to point to `cmsis-svd` repo pull requests.
git filter-repo --replace-message <(echo "regex:^Merge pull request #==>Merge pull request cmsis-svd/cmsis-svd#")
git branch -M filtered-main

# Add new commits from `cmsis-svd-data` since the move on top and clean up.
git remote add origin git@github.com:cmsis-svd/cmsis-svd-data.git
git fetch origin
git switch main
git rebase -i --rebase-merges --onto=filtered-main --root
### Make sure todo content is as follows:
### For a771c3b I changed pick to drop, to remove the previous commit of `data/`.
### For 23899fa I moved it below 2fc4f10 and changed `pick` to `fixup -C` to get rid of 2fc4f10 which only contained an empty README.md
# label onto
#
# # Branch cmsis-svd-add-cmsis-svd-files-from-cmsis-svd-repository
# reset onto
# pick 2fc4f10 Init
# fixup -C 23899fa Add README
# label branch-point
# drop a771c3b Add cmsis-svd files from cmsis-svd/cmsis-svd repository.
# pick a2ff679 Add Apache license.
# pick be27b47 Add CONTRIBUTING
# label cmsis-svd-add-cmsis-svd-files-from-cmsis-svd-repository
#
# # Branch florianhofhammer-add-nxp-mk64f12
# reset cmsis-svd-add-cmsis-svd-files-from-cmsis-svd-repository # Add README
# pick eb72d1c Add NXP/MK64F12 SVD
# label florianhofhammer-add-nxp-mk64f12
#
# # Branch aykevl-espressif-update
# reset branch-point # Init
# merge -C ff486d9 cmsis-svd-add-cmsis-svd-files-from-cmsis-svd-repository # Merge pull request #1 from cmsis-svd/add-cmsis-svd-files-from-cmsis-svd-repository
# merge -C 853bb02 florianhofhammer-add-nxp-mk64f12 # Merge pull request #2 from florianhofhammer/add-nxp-mk64f12
# label branch-point-2
# pick 58fe151 Update Espressif SVD files
# label aykevl-espressif-update
#
# reset branch-point-2 # Merge pull request #2 from florianhofhammer/add-nxp-mk64f12
# merge -C 40327a4 aykevl-espressif-update # Merge pull request #3 from aykevl/espressif-update

# That was all. Now the repo can be pushed, but we changed the history so a force push is required (commented out for now):
# git push -f origin main

# Let's diff against current `cmsis-svd-data` repo, to check that we didn't drop anything.
cd ..
git clone git@github.com:cmsis-svd/cmsis-svd-data.git
diff -r -x .git cmsis-svd/ cmsis-svd-data/
echo $?  # Outputs 0: Folders have identical content
```
I used
```bash
$ git --version
git version 2.43.0
$ git filter-repo --version
ae71fad1d03f
```

VincentDary, 2024-10-25:
@gschwaer  Thank you very much for your proposal, this is very VERY welcome :) , I will review your script today.

@BenBE , @brainstorm, @ckudera, @posborne: This is ok for everybody if I force push here in order to restore the history.

BenBE, 2024-10-25:
IMO, whenever possible the version control should preserve relevant history. So yes, go ahead.

VincentDary, 2024-10-26:
Issue completed, @BenBE thank you very much for your help and creation of the branch and PR for the PR (#5, #6, #7, #8, #9) accidentally closed caused by my force push. 

@rminnich, @ii8, @PfannenHans, @AverseABFun, @zkrx: I am reviewing your PR now.

closed 2024-10-26 by VincentDary