cmsis-svd/cmsis-svd-data #44: [proposal] Synchronise SVD files from official Keil Arm Packs repository

VincentDary, 2024-10-30:
We should consider to synchronise SVD files directly from the Keil Arm Packs repository via the [index](https://www.keil.com/pack/index.pidx) used by `cpackget`. There will be a potential repository size issue to host all the SVD files from the keil repository (to check).

- [cpackget: Open-CMSIS-Pack Package Installer](https://github.com/Open-CMSIS-Pack/cpackget)
- [Keil pack index URL used by cpackget](https://github.com/Open-CMSIS-Pack/cpackget/blob/main/scripts/test-public-index#L9C6-L9C42)

[index_pidx.txt](https://github.com/user-attachments/files/17572716/index_pidx.txt)

erdnaxe, 2025-01-16:
I ran a little experiment by downloading and extracting all SVD from Keil Arm Packs.
As of January 2025, `cmsis-svd-data` size is 5.8 GB. All SVD from Keil Arm Packs is around 9.0 GB.

The pack licence should maybe be considered when synchronizing SVD files?
It seems that most SVD files have some sort of copyright/licencing header.

Observations:
- Some official SVD starts with spaces before the XML declaration, this should get stripped.
- Some SVD files in packs use the `.xml`  file extension rather than `.svd`.
- Some vendor websites are blocking HTTP requests based on IP reputation and HTTP User-Agent.

VincentDary, 2025-01-20:
Hello @erdnaxe, 

Thank you very much for your feedbacks, this is a high value for this future task.

Also, if you have any script to add as a Gist to this issue, in order to help/speed the development, this is welcoming.

erdnaxe, 2025-01-20:
> Also, if you have any script to add as a Gist to this issue, in order to help/speed the development, this is welcoming.

Here is my proof-of-concept: https://gist.github.com/erdnaxe/f680895a3bdd412d760a2e37a2f65365
Feel free to reuse this code.

VincentDary, 2025-01-24:
@erdnaxe Thank you very much for your contribution, this is a good base to start the development of this feature.